kawasaki/pyscrlink #1: SyntaxError: File "./scratch_link.py", line 42

adumont, 2019-11-06:
I've followed the steps in Readme, but I'm getting : 

```
$ sudo python3 ./scratch_link.py
  File "./scratch_link.py", line 42
    logger.debug(f"request: {req}")
                                 ^
SyntaxError: invalid syntax
```

Any idea what is wrong? How can I help troubleshoot?

adumont, 2019-11-06:
Issue is with the f-string syntax. 

In my case, on Ubuntu 16.04.6 LTS:

```
$ python3 --version
Python 3.5.2
```

unfortunately f-strings were added in Python 3.6 only, see [What’s New In Python 3.6](https://docs.python.org/3/whatsnew/3.6.html#pep-498-formatted-string-literals) (here it's called "PEP 498: Formatted string literals").

adumont, 2019-11-06:
seems to work if I call using python 3.7 installed in miniconda.

closed 2019-11-06 by adumont

kawasaki, 2019-11-10:
@adumont Thank you so much for trying on Ubuntu 16.04. Yes, f-string was introduced with python version 3.6. I tried to remove the f-string from bluepy-scratch-link, but found that [websocket module requires python version 3.6](https://pypi.org/project/websockets/). Then, bluepy-scratch-link requires version 3.6 anyway. I will clarify the requirement in README.

Good to know that miniconda helps to introduce newer python version. Wish you can run bluepy-scratch-link with that environment.