kawasaki/pyscrlink #3: Allow local server certificate in Google Chrome

ngammarano, 2020-01-16:
I think a way of allowing a local server certificate in Google Chrome is to type the bypass sequence "thisisunsafe" in https://device-manager.scratch.mit.edu:20110/. This should be required only the first time to access.

jazzpecq, 2020-01-17:
The following patch includes changes to README adding instructions for Chrome browser.

[chrome_support_patch.txt](https://github.com/kawasaki/bluepy-scratch-link/files/4077556/chrome_support_patch.txt)

kawasaki, 2020-01-18:
@ngammarano Wow, I didn't know that special key sequence. Thanks!
@jazzpecq Thank you for the patch! I applied that to the master.

Now I can play micro:bit and Scratch with Chrome :)

closed 2020-01-18 by kawasaki