kawasaki/pyscrlink #5: Bad request :  https://device-manager.scratch.mit.edu:20110

Mijsoot, 2020-02-13:
I think the URL don't work anymore :  https://device-manager.scratch.mit.edu:20110

Have you got some idea ?

Thierry

kawasaki, 2020-02-14:
Hi @Mijsoot,
With my Linux system and home network, bluepy-scratch-link is working well with the URL. Do you observe any error or anything unexpected about that URL?

Mijsoot, 2020-02-14:
[image]

Hi @kawasaki , 
You can what i mean on the attached scrennshot

kawasaki, 2020-02-14:
Hi @Mijsoot ,
Hmm, I see. Then I would like to confirm two things.
Firstly, I assume you already started bluepy-scratch-link with  "sudo ./scratch_link.py" command and that command is running when you saw the error. Is the command running when you saw the error?
Secondly, what happens when you do "ping device-manager.scratch.mit.edu" or "nslookup device-manager.scratch.mit.edu" with you system? With my system, these commands show address 127.0.0.1.

kawasaki, 2020-04-26:
No response for a couple of months. Let me close this issue. If this is still the problem, please open another issue.

closed 2020-04-26 by kawasaki