kawasaki/pyscrlink #9: Scratux support

joancipria, 2020-04-27:
Hi! I'm trying to make it work with [Scratux](https://github.com/scratux/scratux) but it throws the following error when I press the "Try again" button:

> Start session for web socket path: /scratch/ble
> Failure in session for web socket path: /scratch/ble
> code = 1005 (no status code [internal]), no reason

It looks like Scratux is able to connect but then something goes wrong. Any idea about what might be wrong?

[image]

kawasaki, 2020-04-29:
Hi @joancipria   Thanks for the inquiry. Scratux is a good project for kids!

The behavior is weird. When browsers can connect to bluepy-scratch-link websocket server, we usually do not see the "Try Again" dialog. But we see it, and Scratux looks connected to bluepy-scratch-link. The message indicates that the websocket connection is lost between Scratux and bluepy-scratch-link, but does not tell the reason...

Could you edit  scratch_link.py and change log level from logging.INFO to logging.DEBUG? Please do this for both  handler.setLevel() and logger.setLevel(). It will print more debut messages then will help to understand what's going on.

BTW, I tried Scratux on my PC. I did git clone, fetch.sh and npm start. And it worked well with bluepy-scratch-link and my micro:bit. :) So some environmental different between yours and mine maybe the cause of the behavior.

joancipria, 2020-04-29:
Thanks @kawasaki for your answer. I've just discovered that the "Try again" dialog only shows when I have previously enabled my Bluetooth adapter.

With log level set to debug it throws:

> Start session for web socket path: /scratch/ble
> start session hanlder
> start recv_request
> request: {"jsonrpc":"2.0","method":"discover","params":{"filters":[{"services":[61445]}]},"id":0}
> handle request to BLE device
> discover
> {'filters': [{'services': [61445]}]}
> in matches <bluepy.btle.ScanEntry object at 0x7fdfa50d1630> [{'services': [61445]}]
> sevice to check: 61445
> given: 61445
> adtype 128b: None
> adtype 16b: None
> in matches <bluepy.btle.ScanEntry object at 0x7fdfa50d18d0> [{'services': [61445]}]
> sevice to check: 61445
> given: 61445
> adtype 128b: None
> adtype 16b: None
> in matches <bluepy.btle.ScanEntry object at 0x7fdfa50d1710> [{'services': [61445]}]
> sevice to check: 61445
> given: 61445
> adtype 128b: None
> adtype 16b: None
> {'jsonrpc': '2.0', 'error': {'message': "BLE service not found for [{'services': [61445]}]"}}
> response: {"jsonrpc": "2.0", "error": {"message": "BLE service not found for [{'services': [61445]}]"}, "id": 0}
> end_request of BLESession
> start recv_request

But if I turn off the adapter, everything looks right and then it shows "No devices found" (which is true since I don't have a micro:bit). Again, with log level set to debug it throws:

> Start session for web socket path: /scratch/ble
> start session hanlder
> start recv_request
> request: {"jsonrpc":"2.0","method":"discover","params":{"filters":[{"services":[61445]}]},"id":0}
> handle request to BLE device
> discover
> {'filters': [{'services': [61445]}]}
> Failure in session for web socket path: /scratch/ble
> Failed to execute management command 'scanend' (code: 11, error: Rejected)

I think I'm missing something with the BLUE or I'm doing something in the wrong way, especially if it is working for you. Anyway I'm very glad to see it working with Scratux and I would like to integrate it with the project in an easy & fiendly way.

kawasaki, 2020-04-30:
@joancipria Thanks for the debug logs and additional information. Now I see that you don't have micro:bit. If bluepy-scratch-link prints any message when no device is found, it may avoid some confusions. I will think about it.

It will be great if Scratux user kids can play with micro:bit or other devices. Looking forward to it :)

kawasaki, 2020-05-01:
I have added some more log outputs when BLE controllers or BLE devices are not available. It may help users to understand what's going on. Also, I added -d option to scratch_link.py to get debug print easily.

I think this work is done. Close this issue.

closed 2020-05-01 by kawasaki