kawasaki/pyscrlink #10: Support for BLED112 dongle (pygatt)

laurentchar, 2020-05-14:
Hello,

I would like to use bluepy-scratch-link (with scratux) with a BlueGiga "BLED112" (http://www.picaxe.com/BLED112-Bluetooth-USB-Dongle/) Bluethooth USB adapter.

This dongle works with the "pygatt" library (examples found in the "pylgbst" library - https://github.com/undera/pylgbst).
This "pygatt" library is currently not supported in blue-scratch-link.
Is it possible to add support for this dongle?

Thank you.
Best Regards,
Laurent

kawasaki, 2020-05-16:
Hi @laurentchar , thanks for the inquiry.

My understanding is that bluetooth dongle handling is Operating System's job. Especially in Linux system, "bluez" is the sub-system to handle bluetooth dongles. If your Linux system handles BLED112 well, I think it works well with bluepy-scratch-link which depends on bluez and blue-py. Pygatt depends on bluez also, but bluepy-scratch-link not necessary to depends on pygatt, I think.

BTW, if you are interested in Lego Boost support status with bluepy-scratch-link, you may want to check out [this github issue](https://github.com/kawasaki/bluepy-scratch-link/issues/4).

laurentchar, 2020-05-25:
Hi @kawasaki,

The BLED112 works differently than other bluetooth dongles: it does not use the bluetoothctl command but the serial /dev/ttyACM0.

In the meantime I will work with a supported dongle (like the CSR) and help to debug the LEGO Boost support.
I will come back later on to try to add support for BLED112.

Thank you
Laurent

kawasaki, 2020-05-27:
Oh, I totally misunderstood about BLED112.

I found [discussion about BGAPI and BLED112](https://www.silabs.com/community/wireless/bluetooth/forum.topic.html/bled112_device_drive-Ijo1) and [BLED112_Overview.pdf](https://media.digikey.com/pdf/Data%20Sheets/BlueGiga%20PDFs/BLED112_Overview.pdf), and learnt that it has serial device file interface for BGAPI communication and does not communicate via bluez. And pygattt supports the BGAPI communication...  (Thanks, this is educational. I feel python bluetooth support libraries are really diverging :)

I'll focus on Lego Boost as you suggest. Try to find time to work (or play) on weekends :)

miguev, 2020-07-10:
I'm subscribing to this in case I may be able to help with testing; I have a BLED112 dongle, LEGO Boost, EV3, Sbrick and Sbrick+

kawasaki, 2020-07-11:
@miguev, wow, it is a good line up of devices! Once I have something to ask, I will request it. At this moment, BLED112 has lower priority.

kawasaki, 2020-09-20:
I spent some time to investigate about BLED112. It is available on Amazon as of today. It looks BLED112 is popular. It costs around $25. I'm wondering if I can afford it :)  

BLED112 device is supported by python library [pygatt](https://github.com/peplin/pygatt). Introducing this, scratch-link will need to support both two BLE backends, bluepy and pygatt. Such multi-backends will increase code complexity. The python module [btlewrap](https://github.com/ChristianKuehnel/btlewrap/) tries to addresses this multi-backends challenge. It wraps a number of backends and provides single API. If this btlewrap is the stable solution, it would be good to make this bluepy-scratch-link support btlewrap.

I will watch out btlewrap for a while.