kawasaki/pyscrlink #13: Avoid sudo for scratcy_link.py

kawasaki, 2020-07-12:
In issue#4, @miguev pointed out that sudo command is not handy for young children. I open this issue to think about this challenge.

> For young children to use this without depending on adults, it is more convenient to make it work without using sudo with this one-time setup:
> >  $ sudo apt-get install libcap2-bin
$ cd /usr/local/lib/python3.8/dist-packages/bluepy/
$ sudo setcap 'cap_net_raw,cap_net_admin+eip' bluepy-helper
$ /opt/bluepy-scratch-link/scratch_link.py

> I think this would be good to add to the README.md ─ without it, scratch_link.py fails to enable the low energy mode:
> > Failed to execute management command 'le on' (code: 20, error: Permission Denied)

kawasaki, 2020-07-12:
Hi @miguev, thank you again for your comment. I didn't know that setcap command avoids the "sudo btmg le on" command, and "sudo" for scratch_link.py.

One question, in your environment, do you need "sudo btmgmt power on" command before running scratch_link.py? With my Arch environment, still this sudo command is required.

miguev, 2020-07-15:
In my environment (2 PCs, both with Ubuntu 20.04) the command `btmgmt power on` was not necessary. I kept a journal of every step while working on the first PC, so that I could quickly replicate the final working result (out of several tried) in the second PC, and the `btmgmt` command was never used.

kawasaki, 2020-07-18:
@miguev Thank you! I investigated further and learned that if bluetooht.service is enabled and /etc/bluetooth/main.conf has "AutoEnable=true", the service automatically power on the Bluetooth adapter. Actually, these conditions are documented in [ArchWiki Bluetooth](https://wiki.archlinux.org/index.php/Main_page). Then I guess most of distros do not need "sudo btmgmt power on". With your suggestion to set capability to bluepy-helper, I think we can avoid sudo from "./scratch_link.py" command for BLE devices.

Regarding non-LE Bluetooth devices such as Lego EV3, I'm not yet sure if sudo is required for "./scratch_link.py". Do you have EV3 and time to afford to try to connect it using "./scratch_link.py" without sudo?

miguev, 2020-07-22:
@kawasaki running without sudo I was able to connect to the EV3 brick on both systems. However, EV3 requires pairing and this might not work smoothly depending on the desktop environment.

On KDE Plasma, this went smooth because I got a prompt to enter the PIN via a desktop notification.
On XFCE, this took 2 attempts:

- Take 1
Turn EV3 on and try to connect from Scratch, running without sudo.
Scratch finds EV3 brick and I try to Connect to it.
EV3 receives connection request, I accept with PIN 1234 but it fails.
Scratch shows message:
*"Oops, looks like something went wrong."*
There was no prompt for the PIN anywhere.

```
$ ./scratch_link.py
Started scratch-link
Start session for web socket path: /scratch/bt
failed to connect to BT device: [Errno 13] Permission denied
Traceback (most recent call last):
  File "<string>", line 3, in connect
_bluetooth.error: (13, 'Permission denied')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/bluepy-scratch-link/scratch_link.py", line 286, in handle_request
    self.sock.connect((addr, 1))
  File "<string>", line 5, in connect
bluetooth.btcommon.BluetoothError: [Errno 13] Permission denied
```

- Take 2
Open Blueman Manager, Search for devices and Pair with the EV3 brick.
This time, entered PIN 1234 and successfully paired with the EV3 brick. 
Go back to Scratch and click Try Again.
Scratch finds EV3 brick and I try to Connect to it.
Scratch connects to the EV3 brick and I can use it from the Editor.
Again, there was no prompt for the PIN anywhere.

Either way, the whole experience never required to use `sudo` so *it works* in that sense.

kawasaki, 2020-07-24:
@miguev Thank you so much for the trial and detailed report! I think it is good to avoid sudo for non-LE Bluetooth devices also. I will update the README accordingly. I will note that pairing with EV3 brick may be required before connection via bluepy-scratch-link.

kawasaki, 2020-07-25:
In "dev" branch, I have updated the README to avoid sudo from scratch_link.py. Also, I added "setcap.sh" to help to set the capability to bluepy-helper. I will keep the changes in dev branch for a while to review again later.

kawasaki, 2020-08-08:
I merged the changes to the master branch. Done!

closed 2020-08-08 by kawasaki