kawasaki/pyscrlink #15: Lego WeDo Support

KingBBQ, 2020-07-29:
Hello and big thanks also from my side to try to help to support Scratch Link on Linux!

I am working with kids a lot with Lego WeDo 2.0 - the "education" version of boost. Up till now I have been forced to use Windows for working with the devices (as Lego only support Windows and Mac, same for Link) so I was delighted to find your project. 

I was setting everything up according to the docs (I hope I didn't miss anything) and thought I would give it a try. 

When I do, I get this error:

```sh
$ ~/bluepy-scratch-link$ sudo btmgmt power on
hci0 Set Powered complete, settings: powered connectable discoverable bondable ssp br/edr le secure-conn 
$ ~/bluepy-scratch-link$ sudo btmgmt le on
hci0 Set Low Energy complete, settings: powered connectable discoverable bondable ssp br/edr le secure-conn 
$ ~/bluepy-scratch-link$ sudo python3 ./scratch_link.py
Started scratch-link
Start session for web socket path: /scratch/ble
Failed to execute management command 'scanend' (code: 13, error: Invalid Parameters)
Can not scan BLE devices. Check BLE controller.

```

I tried to connect directly via Gnome with the BT-LE device, which works fine. So I guess the hardware should support BLE. 
I am using Ubuntu 20.04

Any help is appreciated!

thanks
Gregor

kawasaki, 2020-07-30:
Hi @KingBBQ thank you for the report. I'm not sure what is the cause of the failure you observe.

I suggest two actions:

1. Run scratch_link.py with -d option. It will prints out plenty of debug logs. Could you save it in a file and attach to this issue? I will take a look in it.
2. There are some improvements in "dev" branch of this bluepy-scratch-link repo. It may worth trying. Especially, if your PC has multiple Bluetooth adapters, a fix in the branch will be required.

KingBBQ, 2020-07-30:
with the Dev branch it works right away!!! great!
tried turning on a motor and that also worked.
let me test a bit more and get back to you!

cheers
Gregor

closed 2020-07-30 by KingBBQ

kawasaki, 2020-07-31:
@KingBBQ Good to know that the issue goes away :)

suoko, 2024-04-13:
I have the same error, so  I tried the dev but I got this below.
Can you suggest something ?
It's wedo 2.0 using Ubuntu 24.04 

2024-04-13 15:05:39,310 loop in BLE thread
2024-04-13 15:05:39,310 start recv_request
2024-04-13 15:05:39,310 loop in BLE thread
2024-04-13 15:05:39,310 loop in BLE thread
2024-04-13 15:05:39,311 start to notify
2024-04-13 15:05:40,598 loop in BLE thread
2024-04-13 15:05:40,598 loop in BLE thread
2024-04-13 15:05:40,598 loop in BLE thread
2024-04-13 15:05:40,599 loop in BLE thread
2024-04-13 15:05:40,599 loop in BLE thread