kawasaki/pyscrlink #18: Delay between sending command from Scratch and forwarding to the device

sza2, 2020-12-14:
Hi,

I developing a firmware and a Scratch 3 extension to make control of Silabs' Thunderboard Sense 2 (BRD4166A) available from Scratch.

I mostly mimic the Micro:bit extension and implemented the firmware based on what I've found in the extension's source code (since I did not find the Micro:bit's firmware code anywhere, it seems only the binary is available).

Te issue is the following: when I click on a block in Scratch which sends a command to the device, there is a big delay between the click and the actual status change on the device.

As I checked, scratch_link.py receives the command quickly (in a millisecond range). The TB Sense 2 device's response also fast - I've tried this directly from Python.

So, it seems, scratch_link.py is the layer which adds the delay. it is in the ~second range (around 0.5s).

The other direction is OK, so reading buttons and tilt values does not suffer from this delay.

Could you please take a look on scratch_link.py and check what causes this delay and whether there is any way to shorten this delay?

Thx!

/sza2

kawasaki, 2020-12-16:
Hi @sza2 , thank you for this feedback! I also wondered why the big delay. Your suggestion and the pull-request looks like a good solution. I don't have time to work on the code on weekdays. Will do at weekend.

kawasaki, 2020-12-19:
@sza2 , I have merged your pull request. Thank you again for fixing this!

P.S. I think some more additional small refactoring may worth, and made in the [dev branch](https://github.com/kawasaki/pyscrlink/commit/1e7f81906e56cbb31102b77b602987d7fdf4be90). Just in case you are interested in and have comments please let me know.

closed 2020-12-19 by kawasaki

sza2, 2020-12-20:
@kawasaki san,

Rather, I should say thank you for pyscrlink! Enabling us to connect Bluetooth peripherals to Scratch ;-)

BR,

/sza2