kawasaki/pyscrlink #20: bluetooth connection to lego boost

pepo-k, 2020-12-21:
Hello,
I have a problem to establish a stable connection to lego boost module.
My set-up:
Lego Boost
+
NUC PC D34010WYK with dualboot ubuntu 20.4 and win 10.

Under Win10 I have installed scratch and also scratch link. There I could get connected stable to lego boost and from scratch drive all motors and also optical interface on lego boost.

Under Ubuntu I have installed scratux and also scratch link (as advised on https://pypi.org/project/pyscrlink/).
I can not say if scratux over scratch link is working at all, as my problem starts at very beginning.
When pairing in bluetooth, I get it on, but in a few seconds later is once again off and the lego boost auto-switches off.
Below the output from terminal while trying to connect. And also the output from hci.
Please help me, I can not find the solution by myself.

    home@ubuntu:~$ bluetoothctl
    Agent registered
    [bluetooth]# pair 00:16:53:A5:14:EC
    Attempting to pair with 00:16:53:A5:14:EC
    [CHG] Device 00:16:53:A5:14:EC Connected: yes
    [CHG] Device 00:16:53:A5:14:EC ServicesResolved: yes
    [CHG] Device 00:16:53:A5:14:EC Paired: yes
    Pairing successful
    [CHG] Device 00:16:53:A5:14:EC ServicesResolved: no
    [CHG] Device 00:16:53:A5:14:EC Paired: no
    [CHG] Device 00:16:53:A5:14:EC Connected: no
    [CHG] Controller D8:FC:93:70:00:B6 Discoverable: no
    [bluetooth]#

    home@ubuntu: hciconfig -a
    hci0: Type: Primary Bus: USB
    BD Address: D8:FC:93:70:00:B6 ACL MTU: 1021:5 SCO MTU: 96:6
    UP RUNNING PSCAN
    RX bytes:2592 acl:30 sco:0 events:163 errors:0
    TX bytes:6262 acl:26 sco:0 commands:124 errors:0
    Features: 0xff 0xfe 0x0f 0xfe 0xdb 0xff 0x7b 0x87
    Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
    Link policy: RSWITCH HOLD SNIFF
    Link mode: SLAVE ACCEPT
    Name: 'pepuntu'
    Class: 0x1c0104
    Service Classes: Rendering, Capturing, Object Transfer
    Device Class: Computer, Desktop workstation
    HCI Version: 4.0 (0x6) Revision: 0xe00
    LMP Version: 4.0 (0x6) Subversion: 0xe00
    Manufacturer: Intel Corp. (2)

kawasaki, 2020-12-21:
Hi @laurentchar, @miguev, @jacquesdt,

@pepo-k is struggling to connect Lego Boost at "paring" step. Could you share any advice?

pepo-k, 2021-01-24:
any progress?

kawasaki, 2021-01-26:
Hi, @pepo-k  Recently I tried scrutux in my environment, and it worked fine with pyscrlink and micro:bit. So, scrutux is not the issue cause, probably.

I don't have Lego Boost and nothing much I can do. One thing I can suggest for you to run scratch_link with -d option with which debug messages will be printed. Please share them so that I can try to find any clue.

grandangelo, 2021-03-28:
Hi guys,

running Debian 11, 4.19.0-9-amd64 kernel, same problem here: I can connect to my Lego Boost but after some seconds it disconnects:

```
andrea@betobeto:~/Python$ bluetoothctl 
Agent registered
[bluetooth]# pair 00:16:53:AA:B0:59
Attempting to pair with 00:16:53:AA:B0:59
[CHG] Device 00:16:53:AA:B0:59 Connected: yes
[NEW] Primary Service (Handle 0x3d5d)
        /org/bluez/hci0/dev_00_16_53_AA_B0_59/service0001
        00001801-0000-1000-8000-00805f9b34fb
        Generic Attribute Profile
[NEW] Characteristic (Handle 0x68ab)
        /org/bluez/hci0/dev_00_16_53_AA_B0_59/service0001/char0002
        00002a05-0000-1000-8000-00805f9b34fb
        Service Changed
[NEW] Descriptor (Handle 0x9da0)
        /org/bluez/hci0/dev_00_16_53_AA_B0_59/service0001/char0002/desc0004
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[NEW] Primary Service (Handle 0x3d5d)
        /org/bluez/hci0/dev_00_16_53_AA_B0_59/service000c
        00001623-1212-efde-1623-785feabcd123
        Vendor specific
[NEW] Characteristic (Handle 0x68ab)
        /org/bluez/hci0/dev_00_16_53_AA_B0_59/service000c/char000d
        00001624-1212-efde-1623-785feabcd123
        Vendor specific
[NEW] Descriptor (Handle 0xa0a0)
        /org/bluez/hci0/dev_00_16_53_AA_B0_59/service000c/char000d/desc000f
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[CHG] Device 00:16:53:AA:B0:59 ServicesResolved: yes
[CHG] Device 00:16:53:AA:B0:59 Paired: yes
Pairing successful
[CHG] Device 00:16:53:AA:B0:59 ServicesResolved: no
[CHG] Device 00:16:53:AA:B0:59 Paired: no
[CHG] Device 00:16:53:AA:B0:59 Connected: no
[bluetooth]# exit
andrea@betobeto:~/Python$ hciconfig -a
hci0:   Type: Primary  Bus: USB
        BD Address: F8:E4:E3:D5:D1:B6  ACL MTU: 1021:4  SCO MTU: 96:6
        UP RUNNING 
        RX bytes:20209 acl:16 sco:0 events:3224 errors:0
        TX bytes:790711 acl:14 sco:0 commands:3202 errors:0
        Features: 0xbf 0xfe 0x0f 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
        Link policy: RSWITCH SNIFF 
        Link mode: SLAVE ACCEPT 
        Name: 'betobeto'
        Class: 0x300104
        Service Classes: Object Transfer, Audio
        Device Class: Computer, Desktop workstation
        HCI Version: 5.1 (0xa)  Revision: 0x261a
        LMP Version: 5.1 (0xa)  Subversion: 0x261a
        Manufacturer: Intel Corp. (2)
```

I don't think it's an hardware issue, because using [this link](https://github.com/ttu/lego-boost-app) I'm able to correctly connect with Boost.

Any advice really appreciated.

kawasaki, 2021-04-03:
Hi @grandangelo

Thank you for the detailed report. I agree that this is not the OS issue. I suppose pyscrlink or its dependent BLE wrapper bluepy has the cause. I want to investigate further, but I don't have Lego Boost. If you can afford the support, it will be appreciated. When you run scratch_link, could you add "-d" option? scratch_link will generated many debug messages. I would like to take a look in that to find out some clue. (After that, I may ask some more debug effort)

grandangelo, 2021-04-03:
Hi @kawasaki,

glad to help you, and thanks for having started this project. About the issue I wrote, I can only guess that it was related to some Debian Testing issue, because I just tried to run pyscratch with "d" option as requested, and it run flawlessy :-) Some update in the last days fixed something...

```
Now I can start Scratch, and while it looks for Boost device I can see in terminal that 
2021-04-03 12:03:39,442 Certificate is ready in FireFox NSS DB: /home/andrea/.mozilla/firefox/f4tttnea.default-esr
2021-04-03 12:03:39,452 Certificate is ready for Chrome
2021-04-03 12:03:39,457 Started scratch-link
2021-04-03 12:03:50,868 Start session for web socket path: /scratch/ble
2021-04-03 12:03:50,868 start session handler
2021-04-03 12:03:50,868 start recv_request
2021-04-03 12:03:50,969 start recv_request
2021-04-03 12:03:50,969 request: {"jsonrpc":"2.0","method":"discover","params":{"filters":[{"services":["00001623-1212-efde-1623-785feabcd123"],"manufacturerData":{"919":{"dataPrefix":[0,64],"mask":[0,255]}}}],"optionalServices":[]},"id":0}
2021-04-03 12:03:50,970 handle request to BLE device
2021-04-03 12:03:50,970 discover
2021-04-03 12:03:50,970 {'filters': [{'services': ['00001623-1212-efde-1623-785feabcd123'], 'manufacturerData': {'919': {'dataPrefix': [0, 64], 'mask': [0, 255]}}}], 'optionalServices': []}
2021-04-03 12:03:50,971 Failure in session for web socket path: /scratch/ble
2021-04-03 12:03:50,971 [Errno 2] No such file or directory: 'getcap'
```

This afternoon I will investigate and see if I can find some solution on my own. I'll share here my findings, if any.

grandangelo, 2021-04-03:
Working!
For some reason were missing symbolic links to getcap and setcap.

I ran as root
```
ln -s /usr/sbin/getcap /usr/bin/getcap
ln -s /usr/sbin/setcap /usr/bin/setcap
```

And now I can happily run my Boost with Scratch :-)

kawasaki, 2021-04-03:
@grandangelo Wow great! I'm very happy you made it! Thanks for sharing the good news :)

In my Arch system, getcap is installed in both /usr/bin and /usr/sbin. Not sure but getcap installation path could be different across distros.

kawasaki, 2021-04-11:
I made a small change in bluepy_helper_cap.py to find out setcap and getcap in /sbin or /usr/sbin even when those commands are out of PATH. Though this would be too much scripting, I wish it will avoid the similar issue that reported in this thread.

With this, the lego boost looks working with pyscrlink. I close this issue. If anyone still has trouble with lego boost connection, feel free to reopen.

closed 2021-04-11 by kawasaki

pepo-k, 2022-01-06:
Hi Kawasaki
I have now tried on another PC to install the scratchlink
https://github.com/kawasaki/pyscrlink#installation
but strugling now on the step#4
What exactly is expected to do there?
If I put it in terminal>
$ bluepy_helper_cap
then> command not found

kawasaki, 2022-01-06:
Hi @pepo-k 
It sounds that "pip install pyscrlink" failing. After pyscrlink installation, bluepy_helper_cap command should be ready. Do you find bluepy_helper_cap in ~/.local/bin? That is the place where pip places commands.

pepo-k, 2022-01-16:
Hi Kawasaki,

it looks like the bluepy_helper_cap is already existing in ~/.local/bin

In the file the following code is in:
#!/usr/bin/python3
# -*- coding: utf-8 -*-
import re
import sys
from pyscrlink.bluepy_helper_cap import setcap
if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw|\.exe)?$', '', sys.argv[0])
    sys.exit(setcap())

After I issue command as normal user:
$ bluepy_helper_cap
I was asked to give sudo password
and after that I got response:
Set capacbility 'cap_net_raw,cap_net_admin' to /home/kubko/.local/lib/python3.8/site-packages/bluepy/bluepy-helper

Even though, I tried to install once again with "pip install pyscrlink" and I got the below response:

$ pip install pyscrlink
/usr/lib/python3/dist-packages/secretstorage/dhcrypto.py:15: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
/usr/lib/python3/dist-packages/secretstorage/util.py:19: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
Requirement already satisfied: pyscrlink in ./.local/lib/python3.8/site-packages (0.2.5)
Requirement already satisfied: pybluez in ./.local/lib/python3.8/site-packages (from pyscrlink) (0.23)
Requirement already satisfied: bluepy in ./.local/lib/python3.8/site-packages (from pyscrlink) (1.3.0)
Requirement already satisfied: websockets in ./.local/lib/python3.8/site-packages (from pyscrlink) (10.1)
Requirement already satisfied: pyOpenSSL in ./.local/lib/python3.8/site-packages (from pyscrlink) (21.0.0)
Requirement already satisfied: cryptography>=3.3 in ./.local/lib/python3.8/site-packages (from pyOpenSSL->pyscrlink) (36.0.1)
Requirement already satisfied: six>=1.5.2 in /usr/lib/python3/dist-packages (from pyOpenSSL->pyscrlink) (1.14.0)
Requirement already satisfied: cffi>=1.12 in ./.local/lib/python3.8/site-packages (from cryptography>=3.3->pyOpenSSL->pyscrlink) (1.15.0)
Requirement already satisfied: pycparser in ./.local/lib/python3.8/site-packages (from cffi>=1.12->cryptography>=3.3->pyOpenSSL->pyscrlink) (2.21)

Do I still miss something?

pepo-k, 2022-01-16:
the tip from grandangelo with symbolic links does not help for me
ln: failed to create symbolic link '/usr/bin/setcap': File exists

grandangelo, 2022-01-16:
Hi pepo-k,

if I run `bluepy_helper_cap` on my system I'm not asked any password, could it be you installed something (maybe via pip) as root?

pepo-k, 2022-01-16:
Hi grandandelo
not that I know about it. 
I was following the manual from kawasaki https://github.com/kawasaki/pyscrlink#installation
pip install pyscrlink
Or if your system has python3 command,
pip3 install pyscrlink
I think in my case was it "pip3".

Beside that I have installed the python with
sudo apt install python3-plp

Can I somehow rework it to start from zero?
Please give me some advises with commands for terminal

kawasaki, 2022-01-18:
Hi @pepo-k 
Your original problem was that you executed bluepy_helper_cap and saw "command not found" error message. Now you are requested to input sudo password, and saw some message "Set capability...". I think you no longer see the error message "command not found", right? If you don't see the error message, it looks that your original problem has got resolved. The message "Set capabiliy..." is the expected correct message.

In case you still want to start from zero, "pip uninstall" command may work. I myself sometimes to do that to clean up my pyscrlink environment. I suggest to do both "pip uninstall pyscrlink" and "pip uninstall bluepy". Then, try "pip install pyscrlink" and "bluepy_helper_cap". Wish it will work for you...