kawasaki/pyscrlink #21: Try to connect to LEGO HUB

rasenderhase, 2021-01-03:
Hi @kawasaki,
I have got a [LEGO HUB](https://www.lego.com/de-de/aboutus/news/2020/june/lego-mindstorms-robot-inventor/) from the LEGO Mindstorms robot inventor set. I am not sure if it should connect aswell using the `scratch_link` script. Is it compatible with LEGO HUB?
I managed to pair the LEGO HUB using bluetoothctl:
```
pi@raspberrypi:~ $ bluetoothctl
Agent registered
[bluetooth]# devices
Device 38:0B:3C:AD:54:5F LEGO Hub 38:0B:3C:AD:54:5F
``` 
Output from scratch_link:
```
pi@raspberrypi:~ $ /home/pi/.local/bin/scratch_link 
2021-01-03 15:34:12,556 FireFox NSS DB not found. Do not add certificate.
2021-01-03 15:34:12,616 Certificate is ready for Chrome
2021-01-03 15:34:12,659 Started scratch-link
2021-01-03 15:34:17,185 Start session for web socket path: /scratch/bt
```
But Scratch is not finding the device. ("No devices found").
Is the LEGO HUB equal / compatible to LEGO MINDSTORMS EV3?
Thanks for your support :-)
Best regards
Andy

kawasaki, 2021-01-04:
Hi @rasenderhase, I walked through the web page of "LEGO Mindstorms robot inventor set". Yes, it will a big fun if we could connect it to Scratch 3.0 :)

My guess is that the LEGO HUB of "robot inventor set" is _not_ compatible to EV3 Intelligent brick.  The appearance is different. LED and display look different also. So, I'm afraid that both official scratch-link and pyscrlnk cannot connect the "robot inventor set" to Scratch 3.0. I found [a discussion thread to wish that in Scratch discussion forum](https://scratch.mit.edu/discuss/topic/458204/). Some robot inventor set owners have same wish as yours, but Scratch 3.0 and official scratch-link do not provide the feature.

In theory, pyscrlink could be modified to convert the messages from "robot inventor set" as if they are from "EV3 Intelligent brick. If we could do this, Scratch 3.0 would be able to control robot inventor set remotely. Said that, It will need certain amount of engineering effort by robot inventor set owners.

rasenderhase, 2021-01-04:
Hi @kawasaki,
thanks for the insights. When I find the time I try to play around with the LEGO HUB. Unfortunately I'm not familiar with Python. I have read - also in the discussion you mentioned above - that it is compatible to a certain boost.
On the other hand it ships with an app that contains a scratch version and that contains the blocks to control the LEGO HUB. Maybe I can find out, how that is working...
Best regards
Andy

rasenderhase, 2021-01-05:
I tried to use the LEGO BOOST library in scratch 3. Now the output of scratch_link is different:
```
pi@raspberrypi:~ $ /home/pi/.local/bin/scratch_link 
2021-01-05 20:43:05,685 FireFox NSS DB not found. Do not add certificate.
2021-01-05 20:43:05,740 Certificate is ready for Chrome
2021-01-05 20:43:05,768 Started scratch-link
2021-01-05 20:43:19,104 Start session for web socket path: /scratch/ble
2021-01-05 20:43:29,321 No BLE device found: [{'services': ['00001623-1212-efde-1623-785feabcd123'], 'manufacturerData': {'919': {'dataPrefix': [0, 64], 'mask': [0, 255]}}}]. Check BLE device.
```
Do you have an idea?

kawasaki, 2021-01-06:
Hi, @rasenderhase , it will be some fun to dig into the LEGO HUB :)

There are two flavors of bluetooth technologies: "Bluetooth Classic" and "Bluetooth Low Energy or BLE". Lego EV3 uses Bluetooth Classic, and Lego Boost uses "BLE". My first question is which of the two the LEGO HUB of robot inventor set uses. I found a description [here](https://www.lego.com/en-us/service/help/products/themes-sets/mindstorms/about-the-lego-mindstorms-robot-inventor-app-408100000020945). It says that the LEGO HUB uses Bluetooth Classic. (It also notes that Bluetooth Low Energy 4.0 also, then it confuses me. Most likely, LEGO HUB uses Bluetooth Classic.) This means that it is the better to try out with Scratch Extension for Lego EV3, than the Scratch Extension for Lego Boost. I think you can confirm that LEGO HUB uses Bluetooth Classic, using the "info" command on bluetoothctl prompt. Please try the command specifying the device address. If the LEGO HUB is a BLE device, some UUIDs for "Generic Access Profile" or "Generic Attribute Profile" will be printed. If LEGO HUB is a Bluetooth Classic device, I expect those UUIDs not printed.

Assuming that LEGO HUB is the Bluetooth Classic device, next question is what is the device class of the LEGO HUB. Looking in [the source code of Scratch Extension for EV3](https://github.com/LLK/scratch-vm/blob/develop/src/extensions/scratch3_ev3/index.js), it specifies majorDevceClass = 8, and minorDeviceClass = 1. These numbers are sent from the Scratch Extension to pyscrlink, and passed to constructor of BTThread class. These device class numbers 8 and 1 are used to find out EV3 Intelligent Brick. To find out LEGO HUB, we need to replace these numbers with the correct numbers for LEGO HUB. My guess is that "info" command of the bluetoothctl can be a help again. It will show "Class" attribute. Please try that command for LEGO HUB, and check the "Class" attribute. Once we know the correct numbers, we can tweak the pyscrlink code to make it find the LEGO HUB.

These are my thoughts about the next steps to dig in. Wish this helps :)

rasenderhase, 2021-11-07:
Hi @kawasaki , it's been a while ;-)

here is the output of info:
```
[bluetooth]# info 38:0B:3C:AD:54:5F
Device 38:0B:3C:AD:54:5F (public)
	Name: LEGO Hub 38:0B:3C:AD:54:5F
	Alias: LEGO Hub 38:0B:3C:AD:54:5F
	Class: 0x00000804
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Vendor specific           (00000000-deca-fade-deca-deafdecacaff)
	UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	Modalias: bluetooth:v0397p0002d0001
```

kawasaki, 2021-11-11:
@rasenderhase Thank you for the info. LEGO HUB has the device class numbers 8 and 4. I will teak the pyscrlink code to make LEGO Hub pretend to be LEGO EV3 (class numbers 8 and 1), hopefully on next Sunday :)

kawasaki, 2021-11-14:
@rasenderhase 

I have done a tweak on the code with the commit 18e0818e9848. It's in dev branch. Can you try that branch on your system? I have no Bluetooth device (I only have BLE device), so the code is completely untested.

rasenderhase, 2021-11-20:
Hi @kawasaki , thanks for all the effort. I replaced the modified script in
```
/home/pi/.local/lib/python3.7/site-packages/pyscrlink
```
Is that correct?

The scratch web site doe not find the device.
[image]

This is the output of `bluetoothctl`:
```
[bluetooth]# info 38:0B:3C:AD:54:5F
Device 38:0B:3C:AD:54:5F (public)
	Name: LEGO Hub 38:0B:3C:AD:54:5F
	Alias: LEGO Hub 38:0B:3C:AD:54:5F
	Class: 0x00000804
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Vendor specific           (00000000-deca-fade-deca-deafdecacaff)
	UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	Modalias: bluetooth:v0397p0002d0001
[DEL] Device 38:0B:3C:AD:54:5F LEGO Hub 38:0B:3C:AD:54:5F
[CHG] Controller DC:A6:32:58:42:63 Discovering: yes
[NEW] Device 47:CB:4D:3E:6C:9E 47-CB-4D-3E-6C-9E
[NEW] Device 65:55:4E:83:CD:03 65-55-4E-83-CD-03
[NEW] Device 38:0B:3C:AD:54:5F LEGO Hub 38:0B:3C:AD:54:5F
[CHG] Device 38:0B:3C:AD:54:5F TxPower is nil
[CHG] Device 38:0B:3C:AD:54:5F RSSI is nil
[CHG] Device 65:55:4E:83:CD:03 TxPower is nil
[CHG] Device 65:55:4E:83:CD:03 RSSI is nil
[CHG] Device 47:CB:4D:3E:6C:9E TxPower is nil
[CHG] Device 47:CB:4D:3E:6C:9E RSSI is nil
[CHG] Controller DC:A6:32:58:42:63 Discovering: no
[CHG] Device 38:0B:3C:AD:54:5F Connected: yes
[CHG] Device 38:0B:3C:AD:54:5F Modalias: bluetooth:v0397p0002d0001
[CHG] Device 38:0B:3C:AD:54:5F UUIDs: 00000000-deca-fade-deca-deafdecacaff
[CHG] Device 38:0B:3C:AD:54:5F UUIDs: 00001101-0000-1000-8000-00805f9b34fb
[CHG] Device 38:0B:3C:AD:54:5F UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device 38:0B:3C:AD:54:5F ServicesResolved: yes
[CHG] Device 38:0B:3C:AD:54:5F Paired: yes
[CHG] Device 38:0B:3C:AD:54:5F Trusted: yes
[CHG] Device 38:0B:3C:AD:54:5F ServicesResolved: no
[CHG] Device 38:0B:3C:AD:54:5F Connected: no
[DEL] Device 47:CB:4D:3E:6C:9E 47-CB-4D-3E-6C-9E
[DEL] Device 65:55:4E:83:CD:03 65-55-4E-83-CD-03
```
and of `scratch_link`:
```
2021-11-20 11:24:02,101 FireFox NSS DB not found. Do not add certificate.
2021-11-20 11:24:02,177 Certificate is ready for Chrome
2021-11-20 11:24:02,193 Started scratch-link
2021-11-20 11:24:10,657 Start session for web socket path: /scratch/bt
2021-11-20 11:24:25,132 scratch closed session
2021-11-20 11:24:59,926 Start session for web socket path: /scratch/bt
2021-11-20 11:25:55,555 scratch closed session
2021-11-20 11:25:55,563 Start session for web socket path: /scratch/bt
2021-11-20 11:26:37,456 scratch closed session
2021-11-20 11:27:03,993 Start session for web socket path: /scratch/bt
2021-11-20 11:31:50,257 scratch closed session
2021-11-20 11:31:50,261 Start session for web socket path: /scratch/bt
2021-11-20 11:32:22,270 scratch closed session
2021-11-20 11:32:23,646 Start session for web socket path: /scratch/bt
```

All this doesn't look so helpful. What else can I provide to you?

Best regards
Andy

kawasaki, 2021-11-20:
Hi @rasenderhase , thank you for trying the modified code.

It is correct that you replaced the file under /home/pi/.local. Another way to try the dev branch is to run pyscrlink/scratch_link.py in the checked out working copy, instead of the scratch_link command. Anyway, your approach should work fine.

I suppose you ran the scratch_link command without -d option, Can you try it with -d option? It will output more messages. I wish it will help me to understand what is going on.

rasenderhase, 2021-11-20:
It seems to go into an infinite loop. However sometimes there "flash" some other lines. I tried to capture them:
```
2021-11-20 19:58:24,941 in handle loop
2021-11-20 19:58:24,941 start recv_request
2021-11-20 19:58:24,943 start to notify
2021-11-20 19:58:24,944 in handle loop
2021-11-20 19:58:24,944 start recv_request
2021-11-20 19:58:24,945 request: {"jsonrpc":"2.0","method":"discover","params":{"majorDeviceClass":8,"minorDeviceClass":1},"id":0}
2021-11-20 19:58:24,946 handle request to BT device
2021-11-20 19:58:24,946 discover
2021-11-20 19:58:24,947 {'majorDeviceClass': 8, 'minorDeviceClass': 1}
2021-11-20 19:58:24,947 Starting async discovery
2021-11-20 19:58:24,948 loop in BT thread
2021-11-20 19:58:24,948 {'jsonrpc': '2.0', 'result': None}
2021-11-20 19:58:24,949 response: {"jsonrpc": "2.0", "result": null, "id": 0}
2021-11-20 19:58:24,951 end_request of BTSession <pyscrlink.scratch_link.BTSession object at 0xb5b65130>
2021-11-20 19:58:24,952 start to notify
2021-11-20 19:58:24,952 in handle loop
2021-11-20 19:58:24,953 start recv_request
2021-11-20 19:58:24,954 start to notify
2021-11-20 19:58:24,955 in handle loop
2021-11-20 19:58:24,956 start recv_request
2021-11-20 19:58:24,957 start to notify
2021-11-20 19:58:24,957 in handle loop
2021-11-20 19:58:24,957 start recv_request
2021-11-20 19:58:24,958 start to notify
``` 

Is that helpful?

kawasaki, 2021-11-21:
@rasenderhase Thank you for sharing the debug log. Though it does not clarify everything, it is definitely helpful. I found two things from it.

1. After the first "loop in BT thread" message, I expect the message "in discovery: state". But it is missing. Session.status may have unexpected value.
2. While I walked through the debug log, I noticed that my code tweak for LEGO Hub was not enough. I modified the handling of device class returned from the device. On top of that, I needed to modify the device class used for device discovery API.

For these two findings, I made another commit 3cf61a145b3a in the dev branch. It adds more debug print to track session status and discovery work, and add one more code tweak for LEGO Hub.

Can you try the dev branch tip?  (I expect that I will need to ask you some more trials based on the results.).

rasenderhase, 2021-11-21:
@kawasaki I wll try it.
I even tried to debug the script with thonny https://thonny.org. But it did not work somehow.
Does the script work on a Mac, too? Then I could use PyCharm. Which IDE do you recommend on a Pi?

kawasaki, 2021-11-21:
@rasenderhase Yeah, you have the device, so your debug can accelerate it, hopefully.

Unfortunately, I'm not familiar with python IDE. I'm a rather old guy, and emacs and print debug with python logger library are enough for me. With quick web search, "Spyder" or "Ninja" would be a good one, but I have no experience on them.

I'm not sure if pyscrlink on Mac either, since pyscrlink is not aimed for Mac :) pyscrlink uses pybluez for BT devices, and pybluez works on Mac. So BT device handling should work on Mac. However, pyscrlink also depends on bluepy for BLE devices which works only on Linux. Certification handling for Linux users may be different on Mac, and it is another factor which makes me doubt about pyscrlink run on Mac.

rasenderhase, 2021-11-22:
```
Print debug messages
2021-11-22 09:45:19,192 set scan_seconds: 10.0
2021-11-22 09:45:19,210 FireFox NSS DB not found. Do not add certificate.
2021-11-22 09:45:19,274 Certificate is ready for Chrome
2021-11-22 09:45:19,290 Started scratch-link
2021-11-22 09:45:28,887 Start session for web socket path: /scratch/bt
2021-11-22 09:45:28,889 start session handler
2021-11-22 09:45:28,891 start recv_request
2021-11-22 09:45:29,004 start recv_request
2021-11-22 09:45:29,006 request: {"jsonrpc":"2.0","method":"discover","params":{"majorDeviceClass":8,"minorDeviceClass":1},"id":0}
2021-11-22 09:45:29,007 handle request to BT device
2021-11-22 09:45:29,008 discover
2021-11-22 09:45:29,009 {'majorDeviceClass': 8, 'minorDeviceClass': 1}
2021-11-22 09:45:29,010 Starting async discovery
2021-11-22 09:45:29,016 loop in BT thread: session status=2
2021-11-22 09:45:29,018 in discovery status:
2021-11-22 09:45:29,018 Search LEGO Hub instead of LEGO EV3
2021-11-22 09:45:29,019 BT discover: class=8/4
2021-11-22 09:45:29,021 {'jsonrpc': '2.0', 'result': None}
2021-11-22 09:45:29,023 response: {"jsonrpc": "2.0", "result": null, "id": 0}
2021-11-22 09:45:29,025 end_request of BTSession <__main__.BTSession object at 0xb514ec90>
```

And another try:
```
...
2021-11-22 09:47:36,965 response: {"jsonrpc": "2.0", "result": null, "id": 0}
2021-11-22 09:47:36,966 Search LEGO Hub instead of LEGO EV3
2021-11-22 09:47:36,969 BT discover: class=8/4
2021-11-22 09:47:36,973 end_request of BTSession <__main__.BTSession object at 0xb5085c90>
2021-11-22 09:47:36,976 start to notify
2021-11-22 09:47:36,978 in handle loop
2021-11-22 09:47:36,979 start recv_request
```

kawasaki, 2021-11-25:
@rasenderhase Thank you for trying the dev branch. According to the log, it looks the pyscrlink is trying to find the LEGO Hub correctly (major/minor = 8/4), but it finds no device. So bluetooth.DeviceDiscoverer API of pybluez does now look working.

My idea of next step is to run simple scripts which use pybluez on your system. Let me have some time to write and try them on my system. pybluez has [some example scripts](https://github.com/pybluez/pybluez/tree/master/examples/simple). I will check if those are useful to test pybluez on your system.

kawasaki, 2021-12-31:
HI @rasenderhase  It's been a while. Finally, I had some time to work on this.

I have attached a simple python script (archived with gzip) to check that LEGO Hub can be detected with PyBluez.
[inquiry.py.gz](https://github.com/kawasaki/pyscrlink/files/7795567/inquiry.py.gz)
If you have time to afford, can you run it on your system and share the output?