kawasaki/pyscrlink #24: Second device disconnected

kawasaki, 2021-04-18:
@shimodash shared a pyscrlink issue through off-line communication today. I record the issue here.

The BLE device [toio](https://www.sony.com/en/SonyInfo/design/stories/toio/) has [its own Scratch enhancement for visual programming tool](https://github.com/toio/toio-visual-programming/), and Scratch-link is required to connect the toio devices and the programming tool. The unique point of the Scratch for toio is that a single Scratch project can connect to two toio devices to allow users to be more creative. The Scratch-link on MacOS and Windows have no problem to support connections to the two toios. However, pyscrlink cannot support it. It is ok to connect the first single toio. When the user started to search the second toio for connection, the connection to the first toio gets lost, and pyscrlink shows the message "Disconnected".

kawasaki, 2021-04-18:
Instead of two toios (I do not have them), I tried to two micro:bit devices to recreate the issue. I opened two browser pages and two Scratch projects in each of them. With pyscrlink running, I established a connection between one Scratch project and one micro:bit. After this, I tried to connect from the other Scratch project to the other micro:bit. At this moment, I observed the connection between the first Scratch project and the first micro:bit was lost, and pyscrlink printed the error message: 

```
2021-04-18 20:21:22,007 Device disconnected
2021-04-18 20:21:22,007 disconnect from the BLE peripheral: BBC micro:bit [zogev]
2021-04-18 20:21:22,019 scratch closed session
```

I guess this disconnection with two micro:bits is the same failure that @shimodash observed with two toios.

shimodash, 2021-04-18:
@kawasaki -san thank you for recording issue!

After off-line communication , I found another problem to connect with toio.
When I put toio on play mat , toio was disconnect to scratch-link with error 13.

log-dump is below

> 2021-04-18 16:47:58,401 BLE notification: 13 b'\x01\xa8\x00\\\x03e\x01\xa0\x00V\x03e\x01'
> 2021-04-18 16:47:58,401 in handle loop
> 2021-04-18 16:47:58,401 start recv_request
> 2021-04-18 16:47:58,402 start to notify
> 2021-04-18 16:47:58,402 in handle loop
> 2021-04-18 16:47:58,402 start recv_request
> 2021-04-18 16:47:58,401 13
> 2021-04-18 16:47:58,402 disconnect from the BLE peripheral: toio Core Cube-D3m

kawasaki, 2021-04-18:
I recreated the issue adding and printing debug messages, and found that the disconnection exception happend in waitForNotifications() call for the first device, just after Scanner.scan() call for the second device completed. These two calls are made to [bluepy](https://github.com/IanHarvey/bluepy). I checked bluepy issue list, and found that this issue is reported multiples times. One of the report [issue number 357](https://github.com/IanHarvey/bluepy/issues/357) was informative. I learnt that the disconnection during scan is the known issue. Fix in bluepy does not seem happening soon.

kawasaki, 2021-04-18:
The workaround idea I can think of is this: let users to make the two devices ready for scan. When the first scan is made, pyscrlink call Scanner.scan() and find the two devices and keep the two devices. When the scan request comes for the second device, do not call the Scanner.scan() and just refer the devices kept.

I did quick dirty code change to test this idea, and it looks working. I will need some more time to have good code. Maybe next weekend?

kawasaki, 2021-04-18:
@shimodash Thank you for the comment, and the issue sharing about the error 13. Not sure, but it looks like that it is another different problem. I would like to focus on the two-devices issue, since I can recreated it in my environment. It will take some more time for me to create a fix trial code. Once it gets ready, I would like to ask your support to try it out.

kawasaki, 2021-05-02:
Hi @shimodash 

I have implemented work around in pyscrlink to allow multiple connections. It's in dev branch of this GitHub repo. Could you try it out? Please make sure to prepare two toio devices ready for scan at the first scan. If you need any support, please let me know.

shimodash, 2021-05-02:
Hi @kawasaki  san

Thank you for telling.
It works fine to support multiple connection!!(awesome)
Two toio is available!
[image]

kawasaki, 2021-05-02:
@shimodash Great! Thank you so much for the quick testing :)
I plan to include the changes to the next release. I hope it will be done within the golden week vacation in Japan.

kawasaki, 2021-05-03:
Hi @shimodash 

I have released the new version 0.2.2 with the multiple device support. With following command, the new package will be in your system, and usual 'scratch_link' command should work for toio.
```
$ pip install -U pyscrlink
```
When you have time, please check it.
Anyway, I think this issue has got resolved. Let me close this issue.

closed 2021-05-03 by kawasaki

shimodash, 2021-05-08:
Hi @kawasaki -san
Sorry reply lately.
I checked new version 0.2.2.
Thank you.