kawasaki/pyscrlink #25: scratch_link procudes endless loop

mslacken, 2021-05-04:
When starting `scratch_link` on openSUSE,  the program crashes and is immediately restarted with the message `Restarting scratch-link...`, what causes an endless loop.
The core of this behavior is that the address `device-manager.scratch.mit.edu` can't be resolved. 
A workaround for this is to add following line to `/etc/hosts`
`127.0.0.1       device-manager.scratch.mit.edu`

kawasaki, 2021-05-05:
@mslacken Thank you for sharing your experience. I will improve the error handling to avoid the endless loop.

I guess this is not openSUSE unique issue. Maybe internet connection is not working good for your system. If internet connection is working, '$ ping device-manager.scratch.mit.edu'  and '$ nslookup device-manager.scratch.mit.edu' should work and show 127.0.0.1. For the systems without internet connection, I think the /etc/hosts edit is the right solution.

mslacken, 2021-05-05:
The internet connection is good, but I doubt that its within the standard, that an address ending with `mit.edu` is resolved to localhost. So I assume that my local ISP does simply does not resolve this address. 
Thanks for handling this.

kawasaki, 2021-05-05:
Oh, I see. Some ISPs could be doing some tricks. My understanding is that device-manager.scratch.mit.edu is the only one domain name among mit.edu to be resolved to localhost as an exception for scratch-link. In my environment, other domain names under mit.edu are resolved not to localhost but to usual addresses.

mslacken, 2021-05-05:
Yes, public ip servers like `1.1.1.1` or `8.8.8.8` also resolve `device-manager.scratch.mit.edu` to `localhost`. Also my the DNS of my local ISP, but the router between drops this address. I assume this is a security feature to avoid the leakage of information of the private network.
I have opened a support ticket at the manufacturer of my router, perhaps they will fix this.

kawasaki, 2021-05-08:
@mslacken Thanks. It sounds more likely that routers do such trick.

I have released pyscrlink 0.2.3 with a small fix for the eternal loop. When device-manager.scratch.mit.edu can not be resolved, scratch_link terminates with an error message.

mslacken, 2021-05-10:
Thanks, so closing this issue.

closed 2021-05-10 by mslacken