kawasaki/pyscrlink #26: BLE session disconnected with toio on playmat

shimodash, 2021-06-05:
Hi kawasaki-san 
I found the issue with playing toio.
I put io on playmat to use absolutely position.
BLE session disconnected was occurred.

I attached log file scratch_link with -d option.
[log.txt](https://github.com/kawasaki/pyscrlink/files/6602015/log.txt)
disconnect may be occurred around line 83705.

kawasaki, 2021-06-14:
@shimodash Hi, thank you very much for sharing the log. And sorry about my slow response. Let me have some time to look into the log and think about it. Yes, the line around 83705 is suspicious. The handle number 13 is the unlucky number :)

kawasaki, 2021-06-20:
I took a look in the log.  Around the line 83705, we can see:

'''
2021-06-05 13:17:07,800 BLE notification: 13 b'\x01X\x011\x01_\x01O\x01,\x01_\x01'
2021-06-05 13:17:07,800 13
'''

These logs indicate that the pyscrlink received notification with handle "13". The first byte of the notification is 1. According to (coreCube.ts)
[https://github.com/toio/toio-visual-programming/blob/master/src/toio/coreCube.ts] of toio-visual-programming, the onNotified method shows that the message with value 1 contains x and y position and direction associated with "Mat". I guess this is the position and direction that toio detected and sent to Scratch.

The weird thing is that the handle 13 is not added to the handle list. At line 43157 and 64070, the log records,

'''
add handle for notification: 39
'''

Then, the handle 39 was registered twice. But there is no log for handle 13.

kawasaki, 2021-06-20:
Hi @shimodash 

For further debugging, I created two commits in the dev branch. If you have time to afford, could you try it? The first commit enriches debug and error message. The second commit is to override the unknown handle 13 with the known handle 39. I'm not sure if this dirty hack works or not, but would like to know what will happen.