kawasaki/pyscrlink #29: Issue during device scanning

advancingu, 2021-11-07:
First of all, thanks for this project to allow using Scratch with Lego hardware under Linux. Unfortunately i have an issue connecting to a Boost device.

---

Ubuntu 21.10
pyscrlink pip3 package version 0.2.4
Firefox 94.0
Lego Boost

Steps I did:
1. Install apt packages `bluez libbluetooth-dev libnss3-tools libcap2-bin libglib2.0-dev`
1. Install `pyscrlink` with `pip3 install pyscrlink`
1. Run `bluepy_helper_cap`
1. Add `device-manager.scratch.mit.edu` as an alias for `127.0.0.1` in `/etc/hosts`
1. Run `scratch_link`
1. Visit `device-manager.scratch.mit.edu:20110` with Firefox and add security exception
1. Stop `scratch_link`
1. Turn on Boost hardware by pressing green button (LED starts flashing white)
1. Use `bluetoothctl` to pair with Boost hardware (`Pairing successful` reported and hardware LED turns from flashing white to permanent blue)
1. Run `scratch_link -d`
1. Go to scratch.mit.edu
1. Start a new project
1. Add Lego Boost block type
1. Click on "Start searching"

Expected result

* Connection to Boost working

Actual result

* Scratch fails to connect
* Firefox web developer tools report WSS network traffic to `wss://device-manager.scratch.mit.edu:20110/scratch/ble` with JSON payload response
  ```
  {"jsonrpc": "2.0", "error": {"message": "Can not scan BLE devices. Check BLE controller."}, "result": null, "id": 0}
  ```
* `scratch_link -d` console output:
    ```
    2021-11-07 16:54:19,006 set scan_seconds: 10.0
    2021-11-07 16:54:19,014 Certificate is ready in FireFox NSS DB: /home/athlon/.mozilla/firefox/[...]
    2021-11-07 16:54:19,019 Certificate is ready in FireFox NSS DB: /home/athlon/.mozilla/firefox/[...]
    2021-11-07 16:54:19,062 Certificate is ready for Chrome
    2021-11-07 16:54:19,073 Started scratch-link
    2021-11-07 16:54:49,592 Start session for web socket path: /scratch/ble
    2021-11-07 16:54:49,592 start session handler
    2021-11-07 16:54:49,592 start recv_request
    2021-11-07 16:54:49,694 start recv_request
    2021-11-07 16:54:49,694 request: {"jsonrpc":"2.0","method":"discover","params":{"filters":[{"services":["00001623-1212-efde-1623-785feabcd123"],"manufacturerData":{"919":{"dataPrefix":[0,64],"mask":[0,255]}}}],"optionalServices":[]},"id":0}
    2021-11-07 16:54:49,694 handle request to BLE device
    2021-11-07 16:54:49,694 discover
    2021-11-07 16:54:49,694 {'filters': [{'services': ['00001623-1212-efde-1623-785feabcd123'], 'manufacturerData': {'919': {'dataPrefix': [0, 64], 'mask': [0, 255]}}}], 'optionalServices': []}
    2021-11-07 16:54:49,696 start BLE scan: 10.0 seconds
    2021-11-07 16:54:49,698 BLE iface #0: Failed to execute management command 'scanend' (code: 13, error: Invalid Parameters)
    2021-11-07 16:54:49,699 start BLE scan: 10.0 seconds
    2021-11-07 16:54:49,701 BLE iface #1: Failed to execute management command 'le on' (code: 17, error: Invalid Index)
    2021-11-07 16:54:49,702 start BLE scan: 10.0 seconds
    2021-11-07 16:54:49,704 BLE iface #2: Failed to execute management command 'le on' (code: 17, error: Invalid Index)
    2021-11-07 16:54:49,704 Can not scan BLE devices. Check BLE controller.
    2021-11-07 16:54:49,704 loop in BLE thread
    2021-11-07 16:54:49,704 {'jsonrpc': '2.0', 'error': {'message': 'Can not scan BLE devices. Check BLE controller.'}, 'result': None}
    2021-11-07 16:54:49,704 send out found devices
    2021-11-07 16:54:49,704 response: {"jsonrpc": "2.0", "error": {"message": "Can not scan BLE devices. Check BLE controller."}, "result": null, "id": 0}
    2021-11-07 16:54:49,704 end_request of BLESession
    2021-11-07 16:54:49,705 start to notify
    2021-11-07 16:54:49,705 in handle loop
    2021-11-07 16:54:49,705 start recv_request
    2021-11-07 16:54:49,705 start to notify
    2021-11-07 16:54:49,705 in handle loop
    [...]
    ```

kawasaki, 2021-11-11:
@advancingu  Thank you for the detailed report. In the console output, the line below is weird.
```
BLE iface #0: Failed to execute management command 'scanend' (code: 13, error: Invalid Parameters)
```
This message comes from bluepy layer. So something wrong happens in that layer. I may need to ask you to get some logs from that layer, but I need some time to investigate how to do that.

In the mean time, please try two actions if you have time to afford. First one is updating all Ubuntu packages to the latest version. In the past, some troubles were avoided with this action. Second one is uninstall bluepy and install it again. It may fix something wrong in that layer.

kawasaki, 2021-11-14:
@advancingu 

I found this error is discussed in [bluepy GitHub issue 150](https://github.com/IanHarvey/bluepy/issues/150). It is said that following command avoids the error on some systems.
```
$ sudo hciconfig hci0 down && sudo hciconfig hci0 up
```
Please try the command above on your system before starting pyscrlink.

In case the command above does not work for you, please check bleutooth service status.
```
$ sudo systemctl status bluetooth.service
```
I found that when that service is stopped, similar error message is reported on my system ("Failed to execute management command 'scanend' (code: 11, error: Rejected)). So I want to ensure that the service is working on your system.
"

Also I found that the similar but different message can be recreated on my system when bluetooth.service stops. 
``` 
Failed to execute management command 'scanend' (code: 11, error: Rejected)
```

advancingu, 2021-11-20:
@kawasaki Thank you for your answers. I finally got around to testing your suggestions.

It appears I was suffering from two issues:

1. The failure to scan. I was able to resolve this by bringing the hci0 interface down and up again as you pointed out.
2. The failure to initiate the BLE connection. It turns out that I must _not_ establish a connection to the Lego Move Hub before starting `scratch_link`. If the Hub is already connected, it does not get found during scanning. So instead, I did pair and trust the Hub using `bluetoothctl` and only then start `scratch_link`. This way the connection got established automatically when searching for the Hub in Scratch.

`bluepy` apparently was fine and I didn't reinstall it.

For documentation, the commands in `bluetoothctl` were as follows:
```
pairable on
agent KeyboardOnly
scan on
# turn on Hub
# look for Lego Hub device ID ($DEVICE_ID below)
pair $DEVICE_ID
# enter number `0` as PIN
scan off
trust $DEVICE_ID
disconnect $DEVICE_ID
```

I will try this out on another PC soon. Hopefully it will work immediately then.

closed 2021-11-20 by advancingu

advancingu, 2021-11-21:
Tested on a second PC. Worked fine right away using the process above.

kawasaki, 2021-11-27:
@advancingu Thank you for sharing your findings :) This github issue itself can be a good reference. I'll think about how to document this in README, or refer this github issue from README.