kawasaki/pyscrlink #30: failed to connect with latest master

vtolstov, 2021-11-13:
```
2021-11-13 16:40:55,134 Start session for web socket path: /scratch/ble
2021-11-13 16:40:55,135 start session handler
2021-11-13 16:40:55,135 start recv_request
2021-11-13 16:40:55,237 start recv_request
2021-11-13 16:40:55,239 request: {"jsonrpc":"2.0","method":"discover","params":{"filters":[{"services":["00001623-1212-efde-1623-785feabcd123"]}],"optionalServices":[]},"id":0}
2021-11-13 16:40:55,239 handle request to BLE device
2021-11-13 16:40:55,239 discover
2021-11-13 16:40:55,239 {'filters': [{'services': ['00001623-1212-efde-1623-785feabcd123']}], 'optionalServices': []}
2021-11-13 16:40:55,246 start BLE scan: 5.0 seconds
2021-11-13 16:41:00,510 in matches 79:09:e5:82:11:7f [{'services': ['00001623-1212-efde-1623-785feabcd123']}]
2021-11-13 16:41:00,511 service to check: 00001623-1212-efde-1623-785feabcd123
2021-11-13 16:41:00,511 given: 00001623-1212-efde-1623-785feabcd123
2021-11-13 16:41:00,511 in matches 62:68:79:c3:db:3e [{'services': ['00001623-1212-efde-1623-785feabcd123']}]
2021-11-13 16:41:00,511 service to check: 00001623-1212-efde-1623-785feabcd123
2021-11-13 16:41:00,511 given: 00001623-1212-efde-1623-785feabcd123
2021-11-13 16:41:00,511 in matches b8:7c:6f:43:2d:af [{'services': ['00001623-1212-efde-1623-785feabcd123']}]
2021-11-13 16:41:00,512 service to check: 00001623-1212-efde-1623-785feabcd123
2021-11-13 16:41:00,512 given: 00001623-1212-efde-1623-785feabcd123
2021-11-13 16:41:00,512 adtype complete 16b
2021-11-13 16:41:00,513 Failure in session for web socket path: /scratch/ble
2021-11-13 16:41:00,513 Error: Non-hexadecimal digit found
```

vtolstov, 2021-11-13:
i don't understand where error happens, "Error: Non-hexadecimal digit found"
I'm try to connect to lego boost

kawasaki, 2021-11-14:
Hi @vtolstov 

The line in the debug output came from BLESession._get_dev_uuid().

```
2021-11-13 16:41:00,512 adtype complete 16b
```

So something strange happened around that function. I have added some more debug prints around that function with the commit 266bcc98bd23 in dev branch. Can you try with the dev branch and share what the debug print output?

vtolstov, 2021-11-14:
thanks, now i have this log
```
2021-11-14 12:52:51,733 Start session for web socket path: /scratch/ble
2021-11-14 12:52:51,734 start session handler
2021-11-14 12:52:51,735 start recv_request
2021-11-14 12:52:51,838 start recv_request
2021-11-14 12:52:51,839 request: {"jsonrpc":"2.0","method":"discover","params":{"filters":[{"services":["00001623-1212-efde-1623-785feabcd123"]}],"optionalServices":[]},"id":0}
2021-11-14 12:52:51,839 handle request to BLE device
2021-11-14 12:52:51,839 discover
2021-11-14 12:52:51,839 {'filters': [{'services': ['00001623-1212-efde-1623-785feabcd123']}], 'optionalServices': []}
2021-11-14 12:52:51,860 start BLE scan: 5.0 seconds
2021-11-14 12:52:57,134 in matches d8:37:3b:8c:a5:84 [{'services': ['00001623-1212-efde-1623-785feabcd123']}]
2021-11-14 12:52:57,134 service to check: 00001623-1212-efde-1623-785feabcd123
2021-11-14 12:52:57,135 given UUID: 00001623-1212-efde-1623-785feabcd123 hash=-6899729616229238819
2021-11-14 12:52:57,135 in matches ed:76:43:08:7c:ac [{'services': ['00001623-1212-efde-1623-785feabcd123']}]
2021-11-14 12:52:57,135 service to check: 00001623-1212-efde-1623-785feabcd123
2021-11-14 12:52:57,135 given UUID: 00001623-1212-efde-1623-785feabcd123 hash=-6899729616229238819
2021-11-14 12:52:57,135 service class uuid for adtype incomplete 16b/2: 0000fee0-0000-1000-8000-00805f9b34fb
2021-11-14 12:52:57,136 uuid: 0000fee0-0000-1000-8000-00805f9b34fb
2021-11-14 12:52:57,136 dev UUID: 0000fee0-0000-1000-8000-00805f9b34fb hash=-7517329118536020616
2021-11-14 12:52:57,136 False
2021-11-14 12:52:57,136 in matches 4a:b0:eb:a7:ae:16 [{'services': ['00001623-1212-efde-1623-785feabcd123']}]
2021-11-14 12:52:57,137 service to check: 00001623-1212-efde-1623-785feabcd123
2021-11-14 12:52:57,137 given UUID: 00001623-1212-efde-1623-785feabcd123 hash=-6899729616229238819
2021-11-14 12:52:57,137 in matches 5c:16:18:5b:f6:90 [{'services': ['00001623-1212-efde-1623-785feabcd123']}]
2021-11-14 12:52:57,137 service to check: 00001623-1212-efde-1623-785feabcd123
2021-11-14 12:52:57,137 given UUID: 00001623-1212-efde-1623-785feabcd123 hash=-6899729616229238819
2021-11-14 12:52:57,137 in matches b8:7c:6f:43:2d:af [{'services': ['00001623-1212-efde-1623-785feabcd123']}]
2021-11-14 12:52:57,137 service to check: 00001623-1212-efde-1623-785feabcd123
2021-11-14 12:52:57,138 given UUID: 00001623-1212-efde-1623-785feabcd123 hash=-6899729616229238819
2021-11-14 12:52:57,138 service class uuid for adtype complete 16b/3: 0000aa01-0000-1000-8000-00805f9b34fb,0000aa02-0000-1000-8000-00805f9b34fb,0000aa03-0000-1000-8000-00805f9b34fb,0000180a-0000-1000-8000-00805f9b34fb,0000fee8-0000-1000-8000-00805f9b34fb
2021-11-14 12:52:57,138 Failure in session for web socket path: /scratch/ble
2021-11-14 12:52:57,139 Error: Non-hexadecimal digit found
```

kawasaki, 2021-11-14:
@vtolstov Thank you for the quick reply! The line below indicates the failure cause:

```
2021-11-14 12:52:57,138 service class uuid for adtype complete 16b/3: 0000aa01-0000-1000-8000-00805f9b34fb,0000aa02-0000-1000-8000-00805f9b34fb,0000aa03-0000-1000-8000-00805f9b34fb,0000180a-0000-1000-8000-00805f9b34fb,0000fee8-0000-1000-8000-00805f9b34fb
```

This line prints out the return value from a bluepy API ScanEntry.getValueText() for adType complete 16b. I assumed that it returns only one UUID, but actually, multiple UUIDs are returned. This is not what I expected.

I have pushed a fix trial commit 3ffbde3 to the dev branch. Can you try it out? Wish it works well.

vtolstov, 2021-11-14:
yes, thanks, i'm try it now

vtolstov, 2021-11-14:
yes, now it connects successful.

vtolstov, 2021-11-14:
now i don't understand why only one motor can works, but may be this is not related to scratch and pyscrlink

kawasaki, 2021-11-14:
Thank you for trying it. One step forward :) Later on, I will work to merge the fix to the master branch.
I'm not sure why only one motor works...

closed 2021-11-17 by vtolstov