kawasaki/pyscrlink #33: BUG Method always returns None and awaits a miracle? ;)

XeNONE1100, 2023-01-20:
Hi, 

Just heaving trying to have some fun with an ev3 and I realized that something is never gonna happen until it's changed...

https://github.com/kawasaki/pyscrlink/blob/751190935a457d7a6a207abb71542188cb76361f/pyscrlink/scratch_link.py#L70

I cannot connect to EV3 as jsonres is always None due to handle_request always returning None. 

https://github.com/kawasaki/pyscrlink/blob/751190935a457d7a6a207abb71542188cb76361f/pyscrlink/scratch_link.py#L80

Log:
`
2023-01-20 11:04:04,499 start recv_request
2023-01-20 11:04:04,499 request: {"jsonrpc":"2.0","method":"discover","params":{"majorDeviceClass":8,"minorDeviceClass":1},"id":0}
2023-01-20 11:04:04,499 default handle_request: discover, {'majorDeviceClass': 8, 'minorDeviceClass': 1}
2023-01-20 11:04:04,499 Failure in session for web socket path: /scratch/bt
`

What should jsonres contain?
Thank you!

kawasaki, 2023-01-28:
Hi @XeNONE1100 

Unfortunately, pyscrlink dropped EV3 support in last July. EV3 uses legacy bluetooth protocol. pyscrlink had used the library pybluez to support the legacy bluetooth sprotocol, but it got unstable. To support EV3 again, I would like to find out other good library to replace pybluez (python-sdbus-bluez looks good, so far). It will take more time to decide good library and re-implement EV3 support.

asaldele1, 2024-03-23:
This pybluez fork works fine with EV3: https://github.com/AcrossTheCloud/pybluez

asaldele1, 2024-03-23:
For now i have created fork that does installs everything for ev3 automatically. It uses dependency link, so you can't use pip install with this repo. To use EV3:
```
git clone https://github.com/asaldele1/pyscrlink.git
cd pyscrlink
python3 setup.py install --user
```