kawasaki/pyscrlink #37: Unable to reestablish connection

acor-uk, 2023-03-31:
Hello,

I have been trying to use Lego WeDo 2.0 with ZorinOS. 

After installation, then running scratch_link, I am able to make connection once using either Firefox and Opera (uses Chrominum engine on Linux?)

Connection is then attempted with Scratch via Browser

I have then tested the connection by running the motor, which works.

After a minute or two of inactivity the device stops responding, despite appearing to still being connected.

2023-03-31 21:47:25,754 Certificate is ready in FireFox NSS DB: /home/myuser/.mozilla/firefox/e26s94rt.default-release
2023-03-31 21:47:25,766 Certificate is ready in Chrome NSS DB: /home/myuser/.pki/nssdb
2023-03-31 21:47:25,795 Started scratch-link
2023-03-31 21:47:32,026 Start session for web socket path: /scratch/ble
2023-03-31 21:47:42,427 connected to the BLE peripheral: Epicman

The device still thinks it is connected, but no response, not recoverable without logout / login or restart. I don't but know how to safely stop scratch_link, so have resorted to CTRL + C, but then this causes another issue when attempting to restart scratch_link

acor-uk, 2023-03-31:
After reading in one of the previous issues, I tried:

```
$ scratch_link -r 2 -s 3
```

Which I then tested using Firefox for a few minutes, then disconnected, then tested with Opera for a few minutes, then disconnected and then tested with the Scratch desktop app (I think is Scratux). All of these worked successfully, but I didn't leave the device to time-out or anything, so will need to check that later.

```
~$ scratch_link -r 2 -s 3
2023-03-31 23:14:52,040 Certificate is ready in FireFox NSS DB: /home/myuser/.mozilla/firefox/e26s94rt.default-release
2023-03-31 23:14:52,643 Certificate is ready in Chrome NSS DB: /home/myuser/.pki/nssdb
2023-03-31 23:14:53,987 Started scratch-link
2023-03-31 23:15:06,011 Start session for web socket path: /scratch/ble
2023-03-31 23:15:09,454 connected to the BLE peripheral: Epicman
2023-03-31 23:19:09,521 scratch closed session
2023-03-31 23:19:09,521 BLE session disconnected
2023-03-31 23:19:09,521 all BLE sessions disconnected
2023-03-31 23:19:09,521 disconnect from the BLE peripheral: Epicman
2023-03-31 23:19:16,764 Start session for web socket path: /scratch/ble
2023-03-31 23:19:28,572 connected to the BLE peripheral: Epicman
2023-03-31 23:19:57,096 scratch closed session
2023-03-31 23:19:57,096 BLE session disconnected
2023-03-31 23:19:57,096 all BLE sessions disconnected
2023-03-31 23:19:57,096 disconnect from the BLE peripheral: Epicman
2023-03-31 23:20:56,638 Start session for web socket path: /scratch/ble
2023-03-31 23:21:00,037 connected to the BLE peripheral: Epicman
2023-03-31 23:21:28,804 scratch closed session
2023-03-31 23:21:28,804 BLE session disconnected
2023-03-31 23:21:28,804 all BLE sessions disconnected
2023-03-31 23:21:28,804 disconnect from the BLE peripheral: Epicman
2023-03-31 23:25:57,873 Start session for web socket path: /scratch/ble
2023-03-31 23:26:01,278 connected to the BLE peripheral: Epicman
2023-03-31 23:31:29,963 scratch closed session
2023-03-31 23:31:29,963 BLE session disconnected
2023-03-31 23:31:29,963 all BLE sessions disconnected
2023-03-31 23:31:29,963 disconnect from the BLE peripheral: Epicman

```

markakis-sch, 2023-04-03:
> then tested with the Scratch desktop app (I think is Scratux)

Instead of Scratux, which is quite an old version, you can use as  desktop application [this build](https://github.com/markakis-sch/scratch-desktop/releases) which is a fork from the official project (with no modifications).

kawasaki, 2023-04-09:
Hello @acor-uk

I read through your report, and still not sure what is the issue. Your first comment reported "After a minute or two of inactivity the device stops responding". Then your second comment noted "All of these worked successfully". So, I guess that you once observed "the device stops responding" issue, but now the issue is not recreated on your system. Is this correct?  I hope so :)