kawasaki/pyscrlink #38: CPU ressources

RonLinu, 2023-07-21:
Hi,

I installed your  'pyscrlink' on Linux Mint. It works fine.

However when I checked in System Monitor, I found out it can use non-negligeable CPU ressource, it can go to 10% and more. And it can reach up to 50% when I stop a Scratch program using my Micro:bit.

I wonder if there is any way to lower the cpu usage? Changing the priority does not help.

kawasaki, 2023-07-30:
Hello @RonLinu  This is interesting. I checked CPU resource using top command while I connect Scratch on FireFox and my micro:bit via pyscrlink. 

```
top - 15:09:54 up 20 min,  3 users,  load average: 2.00, 1.08, 0.56
Tasks: 194 total,   1 running, 193 sleeping,   0 stopped,   0 zombie
%Cpu(s): 27.6 us,  1.6 sy,  0.0 ni, 68.3 id,  1.9 wa,  0.4 hi,  0.1 si,  0.0 st 
MiB Mem :   7814.2 total,   3160.1 free,   3355.6 used,   2315.4 buff/cache     
MiB Swap:   8192.0 total,   8192.0 free,      0.0 used.   4458.6 avail Mem 

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                       
   1452 shin      20   0 3090440 380024 132180 S  76.6   4.7   1:14.33 Isolated Web Co                                               
   1233 shin      20   0   12.3g 840500 359876 S  22.6  10.5   2:21.33 firefox                                                       
   2128 shin      20   0  192372  34316  14364 S  13.1   0.4   0:39.27 python
```

I see the process "Isolated Web Co" uses high CPU resource. Is this same as your environment? According to ps command, this process is part of firefox. The 2nd highest CPU resource eater is another firefox process. And the 3rd process is python, which runs pyscrlink.

I measured CPU usage with Chromium instead of Firefox.

```
Tasks: 217 total,   1 running, 216 sleeping,   0 stopped,   0 zombie
%Cpu(s): 26.3 us,  1.5 sy,  0.0 ni, 71.2 id,  0.2 wa,  0.8 hi,  0.1 si,  0.0 st 
MiB Mem :   7814.2 total,   2300.1 free,   3752.5 used,   3135.2 buff/cache     
MiB Swap:   8192.0 total,   8192.0 free,      0.0 used.   4061.7 avail Mem 

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                       
   3260 shin      20   0 1137.2g 339040 142140 S  81.1   4.2   0:44.34 chromium                                                      
   2901 shin      20   0  266380  36544  14380 S  18.3   0.5   1:26.18 python   
```
Result is similar. Chromium consumes high CPU usage.

So, it looks the CPU resource is consumed on browser processes, not by pyscrlink. I'm not sure what can be done by pyscrlink to reduce the CPU usage.

RonLinu, 2023-07-30:
Yes, Chromium or Firefox can take non-neligeable ressources but when I look in my System Monitor window, I see a specific entry for 'scratch_link'.

I took two screenshots (coudn't figure out yet how to just copy/paste the text ):

This first screenshot is during a connection to a running  Scratch program:

[image]

This second screenshot is after I terminated the Scratch program. This is where it goes a bit crazy. My computer fan starts to spin fast, this is how I found out about this behavior.
I somewhat fixed this with two scripts: one that starts 'scracth_link' and a second one that kills it when I don't need it.

[image]

From what you said, maybe 'scratch_link' is not directly ressource hungry and, instead, inadvertently affect another process.
Ron

deric, 2025-03-02:
I guess the CPU usage could be lower with longer sleep times, e.g. [here](https://github.com/kawasaki/pyscrlink/blob/v0.2.8/pyscrlink/scratch_link.py#L207).

Try using #44, it might use less CPU.