kawasaki/pyscrlink #40: Lego WeDo2.0 not connecting anymore

quamis, 2024-02-16:
Hello,
thanks for you project. My kid loves playing with WeDo2.0 and your project keeps him being able to use it on linux. He used it on an older Linux Mint (originally installed around 2018-2019 i guess) until 1-2 weeks ago. I had the "bright" ideea to upgrade his system to latest Mint version, and now scratch_link doesn't seem to o anything. 

When accessing https://scratch.mit.edu/ and trying to connect to WeDo the connection popup doesn't seem to spend any time searching for the actual device.
In the Firefox console I'm seeing this message:

```
Firefox can’t establish a connection to the server at ws://127.0.0.1:20111/scratch/ble. 
Firefox can’t establish a connection to the server at wss://device-manager.scratch.mit.edu:20110/scratch/ble.
```

Manually connecting to https://device-manager.scratch.mit.edu:20110/scratch/ble returns 
```
Secure Connection Failed

An error occurred during a connection to device-manager.scratch.mit.edu:20110. PR_END_OF_FILE_ERROR

Error code: PR_END_OF_FILE_ERROR
```

As far a I remember PR_END_OF_FILE_ERROR is returned when the https server closes the connection without communicating anything, and scratch_link doesn't seem to log anything regarding to the webserver activity, so i guess that something is wrong on the ws side.

[image]

 
[image]

Chickenbox124, 2024-02-17:
Hay........... I seen that screen before but I don't remember what was the problem

kawasaki, 2024-02-23:
@quamis FYI, the issue #36 reported the error PR_END_OF_FILE_ERROR, and it looks resolved with the action recorded in the issue #35. I hope that it will work for you.

quamis, 2024-02-23:
Hello
installing pip's websockets works.

These are the commands I ran.
```
sudo apt remove python3-websockets
sudo apt purge python3-websockets
pip install websockets

scratch_link -d
```
Thanks

closed 2024-02-23 by quamis

Chickenbox124, 2024-02-25:
IT
WORKS!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

Chickenbox124, 2024-02-26:
🇰🇷

승현 reacted via Gmail
<https://www.google.com/gmail/about/?utm_source=gmail-in-product&utm_medium=et&utm_campaign=emojireactionemail#app>