kawasaki/pyscrlink #42: Can not scan BLE devices. Check BLE controller - Ubuntu 22.04

suoko, 2024-04-14:
I'm trying connecting wedo 2.0 but have this error:

```
2024-04-14 14:22:35,923 Started scratch-link
2024-04-14 14:23:20,919 Start session for web socket path: /scratch/ble
2024-04-14 14:23:21,039 Can not scan BLE devices. Check BLE controller.
```
I started bluepy_helper_cap, I tried "hciconfig hci0 down" and "hciconfig hci0 up", I tried connecting the device with the OS BT pairing first (it does connect), I tried the dev version, but nothing

Can anybody suggest a solution ?

suoko, 2024-04-14:
Also tried:
 sudo setcap 'cap_net_raw,cap_net_admin+eip'  /home/user/.local/bin/bluepy_helper_cap
scratch_link -r 2 -s 3

suoko, 2024-04-17:
I found this node project which works
https://github.com/zdila/scratch-link-ble-linux

markakis-sch, 2024-04-17:
What do you mean "works"? Last year I tried the zdila/scratch-link-ble-linux project with a WeDo 2.0 and all I managed to do was to connect and change the color of the led light. It was impossible to get the motor to work or a sensor to work. What did you manage to do?

Olzaq, 2024-12-26:
I had the similar problem (and IIRC this issue was not visible with this laptop in older distro, but only after an upgrade):

```
2024-12-14 12:01:32,333 Start session for web socket path: /scratch/ble
2024-12-14 12:01:42,474 Can not scan BLE devices. Check BLE controller.
```
You can use e.g. `sudo btmgmt info` to display info about BT controller. For me it show that LE is supported:

`supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr le advertising secure-conn debug-keys privacy static-addr`

But `dmesg` show that LE is not supported (use `sudo dmesg` in some distros):
`[   33.720901] Bluetooth: hci0: HCI LE Coded PHY feature bit is set, but its usage is not supported.` 

Forcing LE on helped:
`sudo btmgmt le on`

And now when starting `scratch_link`:
```
2024-12-14 12:18:06,678 Start session for web socket path: /scratch/ble
2024-12-14 12:18:20,989 connected to the BLE peripheral: BBC micro:bit [tovup]
```
 
So, this is not necessarily an issue with pyscrlink but the problem is with OS settings. Someone who knows more should comment though.