kawasaki/pyscrlink #43: RuntimeError: no running event loop

sponz, 2024-12-23:
Hi,

I am trying to run it on a Linux Mint 21.2, running on an old imac, and I get the following error message:

```
scratch_link -d
Print debug messages
2024-12-23 22:02:53,011 set scan_seconds: 10.0
2024-12-23 22:02:53,011 set scan_retry: 1
2024-12-23 22:02:53,033 Certificate is ready in FireFox NSS DB: /home/admin1/.mozilla/firefox/5srxf0yb.default-release
2024-12-23 22:02:53,058 Certificate is ready in Chrome NSS DB: /home/admin1/.pki/nssdb
Traceback (most recent call last):
  File "/home/admin1/.local/bin/scratch_link", line 8, in <module>
    sys.exit(main())
  File "/home/admin1/.local/lib/python3.10/site-packages/pyscrlink/scratch_link.py", line 576, in main
    start_server = websockets.serve(
  File "/home/admin1/.local/lib/python3.10/site-packages/websockets/asyncio/server.py", line 737, in __init__
    self.server = Server(
  File "/home/admin1/.local/lib/python3.10/site-packages/websockets/asyncio/server.py", line 281, in __init__
    self.loop = asyncio.get_running_loop()
RuntimeError: no running event loop
```

I have followed the instructions, but I am stuck. 

Any help? Kids want to use their legos during Christmas.

Thanks in advance

Olzaq, 2024-12-26:
Hi 
Try this fork: [pyscrlink](https://github.com/FrankFirsching/pyscrlink) which has updated websocket. They should make pull request back here...

In case its helpful here are the steps I did to use it inside python virtual environment: [venv](https://docs.python.org/3/library/venv.html)

Create:
`python3 -m venv pySrcLinkEnv`

Source it:
`source pySrcLinkEnv/bin/activate`

Install suggested package for dependencies (Maybe also works without?)
`pip3 install pyscrlink`

Set capabilites (as root, and make sure to exit from root console)

```
# cd to-users-working-dir
source pySrcLinkEnv/bin/activate
bluepy_helper_cap
exit
```

Remove installed pyscrlink
`pip3 uninstall pyscrlink`

Install build
`pip3 install build`

Build the package (at the working folder where release script is located) 
`./release.sh build`

Install it
`pip3 install dist/pyscrlink-0.2.8.tar.gz`

DancingOnWater, 2025-01-10:
> I have followed the instructions, but I am stuck.
> 
> Any help? Kids want to use their legos during Christmas.
> 
> Thanks in advance
I have the same troble. The problem is in version websockets. Workaround - downgrade to webscokets-13.

futzlig, 2025-03-03:
Hello @Olzaq ,
with your patch scratch_link seems to start on my machine but Scratch still says it's missing scratch link. I log at other topics here and all log output have a line "scratch link started". On my output I have two lines the Certificate ist ready for Firefox and Chrome and that's all. Am I missing something?
Bluetooth is active, I can connect to my Wedo Device using Ubuntus bluetooth settings dialog.

heepdog, 2026-01-28:
I also just downgraded websockets  `pip install websockets==13` inside my virtual env, ran `bluepy_helper_cap ` and I was able to get it connected.  It was a little fiddley, seemed like you had to turn on the WeDo after you searched for it, or had to do it twice.  Also after disconnecting it, scratch_link needed to be restarted if I wanted to reconnect with the WeDo